smagicld/xiaoxiangBMS #1: Do not find any BMS

kgabor1, 2018-05-11:
Hello,

I have 3 Xiaoxiang BMS with bluetooth module, and it works fine from Android. After trying your app it does not find any BMS. When I start searching with Android, the led blue light turns on instantly on BMS module, but with this app it does not happen. I have iPhone 6S, iOS 11.3.1.

smagicld, 2018-05-18:
Hi

I've testet the app with the Xiaoxiang BMS and Bluetooth Module from 2017. Is it possible for you to make a picture of the Bluetooth Module you're using? (I need to know the chip type) I'll try to get it and implement.

smagicld, 2018-05-18:
Hi Again

Another option: please provide me the ServiceUUID of your Xiaoxiang bluetooth modules.
I've described it here how to do this: [ServiceUUID Guide](https://github.com/smagicld/xiaoxiangBMS/wiki#get-your-bluetooth-module-information)

I'll build you an beta version for testing.

smagicld, 2018-05-20:
UPDATE: I could evaluate the ServiceUUID of the old Bluetooth Modules.

If you post me your E-Mail address, I can add you as beta user to test the latest release.
Let me know.

smagicld, 2018-05-26:
Hi

I've released version 1.0.4 that shows older bluetooth modules. can you please confirm me it it works for you? Unfortunately I do not have the old bluetooth module to test it myself...

closed 2018-06-05 by smagicld