smagicld/xiaoxiangBMS #2: Rename BMS

cdc31, 2018-05-26:
Hi,

would it be possible to add the option to rename the BMS, so that several different ones don't all show up as "xiaoxiang BMS"?

If it's not possible to actually rename the device itself (can you?), then maybe it would be possible to assign names inside the app for each BMS?

Thanks
Clemens

cdc31, 2018-05-26:
I see, would it be possible to just associate a name in your app with a particular BMS/BT device without renaming the physical device? Do they have unique identifiers?

Thanks!

smagicld, 2018-05-27:
Hi Clemens

I see your problem - i have 3 BMS's at home and always guessing which one is which. But the Idea with UUID is very good! Till next Wednesday i'm in Shanghai and have no access to the Hardware, but as soon I'm back - i'll check it.
Idea:
1. Main screen lists visible BMS. For every found BMS an local lookup will be made: UUID -> Stored name.
2. On the BMS Screen, it will be possible to set the name. I'm thinking to add the option to rename the BMS module name too (same as in the Desktop application).
3. Once the name is set - store the UUID -> Name mapping.

I'll try it to implement in 1.5 Week.

smagicld, 2018-06-04:
Hi Clemens
I've found out, that each Bluetooth Module has its own identifier.
It will be possible to add alias names for the batteries!

I'll try to implement it this week.

Best,
Lukas

Example output of 3 different batteries:
didDiscover: <CBPeripheral: 0x1d4118a50, **identifier = 2B737CD3-4239-2D37-C75E-16677591C524**, name = xiaoxiang BMS, state = disconnected>, rssi: -91
didDiscover: <CBPeripheral: 0x1c8115060, **identifier = EE0FBEB9-CC26-2266-9117-31C4AF3AF47A**, name = xiaoxiang BMS, state = disconnected>, rssi: -89
didDiscover: <CBPeripheral: 0x1d011ac10, **identifier = BB885450-778A-A1CF-1A96-A093F4631992**, name = xiaoxiang BMS, state = disconnected>, rssi: -91

smagicld, 2018-06-04:
Clemens
If you want to test this feature in the latest beta version, please give me your e-mail address. I'll add you to the test group.

cdc31, 2018-06-04:
Hi, sure, that would be great. I didn't find an option to message you my e-mail privately though... :/

smagicld, 2018-06-05:
In the App go to Info. There is my email address. just drop me an email.

smagicld, 2018-06-06:
Hi Clemens

Today you should get invitation from apple TestFlight. Please install the App (v 1.0.5) and let me know if you're happy with the solution 😄

BR,
Lukas

cdc31, 2018-06-06:
Thank you! Love it, makes it so much nicer to see real names for the various batteries!

But I would suggest you move the config button somewhere to the bottom, as right now it interferes with the nice design at the top. It’s not symmetric anymore... :)

smagicld, 2018-06-06:
Glad you're happy 😄

I've moved the button to bottom-right corner.
I'll add there another button to switch ON/OFF the output and input MOSFET.

[image]

The Issue will be closed.

closed 2018-06-06 by smagicld

desultorynz, 2019-02-06:
Did you find any way to actually rename the bluetooth / bms name, so if you are viewing on multiple phones / devices then you can see the same name on each to know which battery / device you are looking at?

RobDuthie, 2019-06-16:
> 
> 
> Hi,
> 
> would it be possible to add the option to rename the BMS, so that several different ones don't all show up as "xiaoxiang BMS"?
> 
> If it's not possible to actually rename the device itself (can you?), then maybe it would be possible to assign names inside the app for each BMS?
> 
> Thanks
> Clemens

Hi 
Just installed a the new BMS  BT module and you can change the password and name now to suit your device. On the Xiaoxiang BT BMS module

Bangybug, 2021-07-26:
I want to use HM-10 bluetooth module with the BMS. Do you think it is sufficient to set 'xiaoxiang BMS' name in my Bluetooth module?

smagicld, 2021-07-26:
Yes, you can use any Bluetooth supporting RS232. Important is to connect correctly RX and TX. 
you can use any name for the bms. Important is to set the same Bluetooth Advertisement / Service ID‘s in order to be detected as Xiaoxiang compatible BMS.