smagicld/xiaoxiangBMS #5: Show an overview for all BMS/Packs

cdc31, 2018-06-06:
An idea for future releases...

Right now you show the signal strength and the BMS name on the front page.

It would be really cool if you had the option to show it like this:

BMS_name1      capacity_%       battery_icon
BMS_name2      capacity_%       battery_icon
Etc.

This would make it really easy to select a battery and so final checks etc.

The method I can see is that you could just connect to each BMS you find and read the capacity, then connect to the next. Do this while showing ...3/5... or so, then show all the results in an overview.

smagicld, 2019-10-15:
Is this what you are asking for? 😄

[image]

Jaco-TTT, 2022-07-22:
> Is this what you are asking for? 😄
> 
> [image]

If you can add the Delta there too, it would be perfect.

smagicld, 2022-07-22:
delta of what? cells?

Jaco-TTT, 2022-07-22:
> delta of what? cells?

Yes, the Delta of the cells overall, as you have in the App already. 

The reason is it is a quick glance to see which bank needs some attention i.e. a bank's Delta exceeds i.e. 0.100v to go and check which cells need attention.

Jaco-TTT, 2022-07-22:
To confirm, this Delta on the front screen too.
[image]