smagicld/xiaoxiangBMS #6: Charge/discharge with power on/off

adilism1, 2018-06-15:
Hi again,

So I checked the updated version on/off switch and it works perfectly. One thing I think will be very useful is to allow charging, when the discharge is switched off. That will allow to charge the vehicle, with all other parts switched off. Also, I wanted to check with you if it is safe to charge the vehicle with a connected laptop charger, when the bms is switched on, without disconnecting the ESC and motor? Please let me know. Aside from that, the app works very well.

Best regards,
Adil

smagicld, 2018-06-16:
1. It is my case too. I need to switch off the output leaving charging port on. I'll implement an config element to leave the charging port on while the battery is off.

It will come with next release.

2. Yes, i charge all my equipment (laptop too) while charging the batts from solar. if it is safe - it depends only from your charger. the most important thing is to make sure, your charger accepts the whole voltage range of your charger. for example, if your charger accepts strict 12v on input, it should be not used in car - the voltage can goes up to 14.8V.

br,
lukas

smagicld, 2018-06-16:
I have implemented this setting param as described before.
You should get tomorrow beta version (1.0.7) over test flight. 
let me know if it works for you too.

[image]
[image]

closed 2018-09-16 by smagicld