smagicld/xiaoxiangBMS #13: Modest locked

jdemourgues, 2018-10-19:
The MOSFET Software is locked.
How to unlock it to run my scooter
Thank it is urgent
Jerome

smagicld, 2018-10-19:
Connect to bms and press the green „batt on“ Button.

closed 2018-10-19 by smagicld

jdemourgues, 2018-10-19:
It does not work when I press "Batt on" on button.
On the Android App the red lock button stays.
I did try to lock the battery with a password ... It locked it but now, impossible to unlock it ..
And I cannot go back home 
Thank you for helping me
I neither cannot read the data on the bms ... as it is locked

smagicld, 2018-10-19:
I‘ll send you an Dropbox link with windows software tomorrow (i‘m not at home now). Just leave me your email.

jdemourgues, 2018-10-19:
Thank you very much.
You can send it on jerome[a]demourgues.com
Best regards,
Jérôme

jdemourgues, 2018-10-19:
Do you know the initial PassWord on the android app and the BMS and how reset the BMS to factory defaults

smagicld, 2018-10-20:
here is the link to windows software to manage the BMS
https://www.dropbox.com/s/9nwvbolybz44g5d/parameter_set_tools.zip?dl=0

smagicld, 2018-10-20:
what is the exact model of your bms? if you do not know it, can you take a picture of the labels on it?

jdemourgues, 2018-10-20:
Hi,
The BMS is a JBD-SP15S003 v1.1

jdemourgues, 2018-10-20:
I try to send you a picture with the mail