smagicld/xiaoxiangBMS #15: Andoid version

quadlike, 2018-11-02:
Hello, very interested in your project. I am amazed at the functionality that you managed to put into the mobile application.
Do you plan to release a version for android? I think the demand for it will be high, the original application sucks. With your application, there is no need to connect the BMS to the computer ... respectively, and the UART does not need to buy. It costs about $ 5-10. I am ready to pay this money for the application

heruamart, 2018-11-07:
Hello.
The android community needs your project

vladanstrigo, 2018-11-29:
I would be also very interested in Android version. Willing to pay as quadlike as well.

mamuka2006, 2018-12-15:
I welcome, I really miss your program for android, please make ready to buy! Thank!

thanarious, 2019-05-03:
There are a couple of applications for android that support such BMSs, most are also free. However, they seem to be focused on e-bike / EV applications, so they come with all the extra bells and whistles such applications require, e.g. speed reading and stuff.

Ovaday, 2019-05-03:
But it would be really nice if developer made an Android app. The original had some bags, I know critical one that could damage our bms after sequence of actions.
I want to buy the app, that really would be supporting by person with normal hands.

quadlike, 2019-06-02:
thanarious
what applications are you talking about?  Those that are in the play market do not support this bms

smagicld, 2019-06-02:
I'll take a look on porting the iOS App into Android world. But first I'll have to collect some money to get Android devices for development. Unfortunately the communication with BMS is very low level implementation and can not be virtually simulated.

thanarious, 2019-06-07:
AFAIK, [this android application](https://www.lithiumbatterypcb.com/APP%20Software%202.1.1024.zip) does support a couple of my smart BMSs.

garciasa, 2019-06-12:
My two cents, @smagicld port your app to flutter, and you can keep two version effortless.

kubbur, 2019-11-20:
Any news on this ?

mgrn0, 2021-04-02:
I would be also very interested in an Android version.