smagicld/xiaoxiangBMS #19: Balance when charging AND not charging

anburton, 2018-11-23:
when I select Balance when chargingnonly it does what it says and balances when charging only.  when I unselect it, it only balances when it stops charging due to an overvoltage condition (single cell maxed out), and balances when its bms alarm shuts it off.   When it balances a cell down to 4.150 v or so, it charges again,  the cell trys to charge back up rather than getting balanced when charging up.  can It balance a cell during BotH charging and when its idle balancing due to single cell overvoltage?

smagicld, 2018-11-23:
It depends on firmware of your BMS. you have to ask directly your reseller.

closed 2018-11-23 by smagicld

anburton, 2018-11-23:
This is what i purchased on AliExpress
https://s.click.aliexpress.com/e/Kpatdvq

You mean that the way balancing charging works is FW controlled?  Do you work with the HW mfg of these Aliexpress smart BMS?  I wonder if they have access to this if I share the link in Aliexpress...  ok... can I upgrade FW from your app?

smagicld, 2018-11-23:
Firmware can be updated only with the original windows software.
No, I do not work with the manufacturer or any ali reseller - they do not want to cooperate.
the whole app is rev-engineering.