smagicld/xiaoxiangBMS #22: data logging

BreitiDE, 2018-12-27:
log data per second, ability to select what’s to be logged, ability to save and send logs, simple log structure with fixed length or csv

BreitiDE, 2019-02-17:
save logs in icloud drive would be perfect

thanarious, 2019-02-20:
I just spent a couple of hours searching for the logs; is this feature implemented, or not yet? Indeed, using iCloud Drive would be great, it would automatically work as a cloud BMS service!

victornorberg, 2019-05-23:
+1 for logging

smagicld, 2019-08-19:
Hi All

I need your feedback to the Logging feature:

1. In BMS Settings you'll have the option to enable file logging. It will also be possible to define the filename. Default: logging_BMSalias.log

2. New Records / measurements will be written every 0.5 sec.

3. The file format will be like this:
- per line one record (enden with new line character)
- each line contains an JSON object with all known Field. This is better as CSV because the structure can be easy extended in future very easy and we have no formatting problems. JSON Object is also very simple to parse.

OR you say JSON is no go - you want to have CSV?????

4. The written LOGs can downloaded over iTunes File Share.

5. There will be no additional UI for analysing recorded data. Reason: Charts / analysis can be done much easier / faster on desktop.

What do you think?

thanarious, 2019-08-20:
I believe you should ditch local storage and opt for iCloud, since that way it will be online and we'll be able to parse it in real-time from another device. I personally don't care about the data format yet, since I believe getting to the data is much more important. Also, being able to adjust write interval would also be a good idea, maybe 0.5s is a bit too often for any real-time monitoring.