smagicld/xiaoxiangBMS #23: iPad compatibility

BreitiDE, 2018-12-30:
it would be great if I can use the App on my iPad in landscape mode. I use one iPad as central control system which is powered on the whole time. this make it fantastic for logging and sharing the charging curve

thanarious, 2019-01-23:
I was surprised to see an iOS version for these BMSs. On the matter of compatibility, is there any way to lower the iOS version requirement? Is there a particular reason that iOS 10 is required for this application to work?

smagicld, 2019-01-24:
Logging is planned, but can not tell you when I will release it (probably in 2 months).
IOS10 is the minimum I can build - because of Apple Watch compatibility. and I do not want currently to have two different apps in the store.

smagicld, 2019-07-03:
iPad support implemented.

closed 2019-07-03 by smagicld