smagicld/xiaoxiangBMS #24: Static Balancing enable / discable for ios ?

desultorynz, 2019-01-07:
Good evening, I was just speaking with one of your support people and they advised me to TURN OFF "Static Balancing" so the bms doesn't balance when the battery is idle (but rather only when charging), but they've advised me that there's no current way to DISABLE static balancing from the current IOS app sadly.  Even though I have Balancing mode enabled, and then Balancing only when charging enabled, the bms still wants to balance on it's own even when the battery is totally disconnected (eg nothing bolted to the terminals).  Are you able to please get the static balancing enable / disable function added into your app thanks.

smagicld, 2019-01-07:
It is already there!
Go to BMS -> config. then BMS Settings Tab. The option is: "Bal. only when charging". If this is switched on, balancing will be on only when charging.

closed 2019-01-07 by smagicld

desultorynz, 2019-01-07:
Actually the BALANCING ONLY WHEN CHARGING option is turned on, however it still balances even when the battery is sitting disconnected (no charging leads connected to the battery at all), so clearly there's an issue there where I now can't turn off the static balancing with the ios app :(

desultorynz, 2019-01-07:
I even tried turning ALL balancing options off and then hit WRITE, and it said success, and then went back to the main status screen and it still showed each cell balancing (without anything connected to the battery terminals at all).

smagicld, 2019-01-07:
To prevent misunderstanding I use exactly the same naming as in the Windows App / official windows App documentation.

desultorynz, 2019-01-07:
I see, no problem, I just did not realise that Light Grey Balance word actually means balance is off (not currently active working), and dark grey means it's on / currently active working.