smagicld/xiaoxiangBMS #25: Password / Pin needed to adjust / write settings?

desultorynz, 2019-01-07:
Is it possible you can make it so I need to enter a password / pin into the app to be able to get into advanced mode to view and write config settings, but so that end users who are using the app can just see more basic data such as cell voltages + balancing + battery voltage + battery soc % for example but without all the other technical stuff to mess up?

For ios app, thanks.

smagicld, 2019-01-07:
I'm working on it now. 
In the coming release only for accessing PIN. 
Writing will become an In-App purchase, so I think normal user will not want to buy this option.
If this will be still an issue, I'll add pin explicitly for writing (in case write option in In-App was purchased).

closed 2019-01-07 by smagicld

desultorynz, 2019-01-07:
Thanks, yes if you can please choose to require a pin first for ALL writing function, that would be good, just in case a user does purchase the app, so they don't mess up critical battery settings.  Or else even have a LITE version of the app which simply only displays basic end user data such as SOC % etc.  So to not confuse end user customers with too much technical data, yet still allow experts the ability to access this with a pin.