smagicld/xiaoxiangBMS #26: Ability to save a setting configuration template?

desultorynz, 2019-01-07:
Can you please make it so I can create and save a default setting template for all config settings, so that if I have many batteries I need to configure then I can simply load the config, connect to the BMS and then write the config to that bms, rather than needing to manually enter each setting for each bms manually, thanks.

BreitiDE, 2019-01-08:
yo, this is a cool idea, what BMS are you using ? I got this one from Ali http://s.aliexpress.com/IZveaUvm

desultorynz, 2019-01-10:
We use various versions of the Smart BMS including the 7S, 10S, 16S models etc, various sizes / current options also depending on the battery build configuration and current output required etc.

smagicld, 2019-05-16:
working on it...

smagicld, 2019-05-29:
I think it is done. you can save and open configs as in the desktop tool.
Testflight users can test it now. If no big issues, AppStore release should come next week.

closed 2019-05-29 by smagicld