smagicld/xiaoxiangBMS #27: Max Charge / Discharge current setting ?

desultorynz, 2019-01-07:
Can you please add a max charge + max discharge current setting, so if we have a 100Amp bms we can manually set the charging to say 50Amp, and discharging to say 80Amp, to better look after the cells in the battery thanks.  The app doesn't currently allow us to do this sadly.

thanarious, 2019-01-23:
Afaik, a BMS can only disconnect the battery, it cannot dynamically regulate charge/discharge current.

thanarious, 2019-01-24:
By the way, if you want to limit charging current, you better do it through the charger, because if the BMS cuts off, this could damage the charger.

smagicld, 2019-05-08:
Fields: 
- Charge Overcurrent Protection (CHGOC)
- CHGOC Release and Delay
- CHOGC Delay
- Discharge Overcurrent Protection (DSGOC)
- DSGOC Release and Delay
- DSGOC Delay

Same for Cell under/over protection and whole pack under/over protection.

are now implemented.

If you are not in the testflight group just give me your email - you can test the beta.

closed 2019-05-08 by smagicld

thanarious, 2019-05-09:
How does one enter the testflight group?