smagicld/xiaoxiangBMS #29: Calibration documentation

thanarious, 2019-01-24:
I am having issues with current readings on my BMS. Seems like both charge and discharge currents as reported by the BMS are off by around 40%. Is this normal? Can this be corrected by the calibration settings of the app? What are the basic guidelines for current calibration, given the caution notice in the app?

smagicld, 2019-01-24:
The App itself shows values that BMS is reporting. It depends on your BMS / seller if he calibrated it before or not. It look like it is not calibrated. How to do this: first calibrate 0 (without any charge or discharge). then connect discharge, measure the current and enter it into the App as the right value.
If you need more details for calibration refer to the manual you have got with the BMS or ask the shop where you have bought the BMS.

closed 2019-01-24 by smagicld

thanarious, 2019-01-24:
Thank you for the information

thanarious, 2019-01-24:
Dear Lukas,

If I calibrate the BMS current at -say- 3A and then find out that it’s off at 0.5A and I go ahead and calibrate again at 0.5A, will that keep both calibrations? Because I am still seeing that it veers off at different currents...