smagicld/xiaoxiangBMS #30: Missing parameters?

thanarious, 2019-01-24:
I am using and android version from a BMS vendor that has lots more parameters that can be viewed and changed:

Overvoltage, release and delay
Undervoltage, release and delay
Pack overvoltage, release and delay
Pack undervoltage, release and delay
Charging/discharging low/high temp, release and delay
Charge/discharge overcurrent, release and delay
Galvanometer resistance
Hardware overcurrent protection and delay
Hardware short circuit protection and delay
Hardware overvoltage protection and delay
Hardware undervoltage protection and delay

Next, there’s the parameters that are included in the iOS application as well

I would also like to ask what the bottom 5 settings are.

desultorynz, 2019-03-01:
I too agree it would be super helpful to be able to adjust these items as soon as you can avail thanks.

smagicld, 2019-05-08:
is now in beta testing. all current and voltage under/over protection fields will come with next release (in a week / two)

closed 2019-05-08 by smagicld

thanarious, 2019-05-14:
Thanx for including me in the beta testing version. Please elaborate on the "Functional Configs": Switch, SCRL, LED_EN and LED_NUM, I still have no idea what they mean!

smagicld, 2019-05-14:
I'll add short description of it. This is what I've got from my contact person:
[image]

to the LED: this is future reserved flag. the prototype board has them:
[image]

[image]