smagicld/xiaoxiangBMS #31: Off-Topic: Revitalise nonworking JBD-BLE001 / TLSR8266

careyer, 2019-01-24:
Hi @smagicld,

bought your amazing App in the AppStore and wanted to try it with my brandnew SmartBMS. Unfortunatly the BMS is not found via bluetooth. Obviously the seller (AliExpress) did not programm the BT module correctly. It blinks in blue but "BlueCap" and "LightBlue" can not find any new bluetooth device. Do you have a hint on how i can flash the module accordingly to make it work or point me to the right direction?

Thanks!
Thomas

BreitiDE, 2019-01-24:
Hi Thomas, I am not sure, but it can be that you need to rename the bluetooth module using serial terminal ...

careyer, 2019-01-24:
Hi @BreitiDE - thanks for your quick answer! How would I do that? Connecting the module via FTDI and see what it does on Putty? What also bothers me is: My BMS seems to have the full battery voltage also on the 4-Pin Dataport? For testing purpose I connected the BMS to my bench powet supply (12V) and when measuring between VCC and GND (on the data port) it reads 10.8V... This is very unusual isn't it?

BreitiDE, 2019-01-25:
Hi @careyer I currently have all my BMS housed, but waiting for a new one ... as I know the bluetooth module is driven by 3 - 5 V. And yes, you can found some hints on google to set the bluetooth speed and name depending on the module manufacturer.

careyer, 2019-01-25:
@BreitiDE can you give me any hint were to search on google? I tried googling for "TLSR8266 programming" or "flashing TLSR8266" but without success

p-f-w, 2021-05-04:
newer(?) BLE-Modules have TLSR8251 in 24 or 32 pin layout, they can connect via the UART-USB-adapter (from JBD), i ordered few adapters separately which comes with extra cables, one of them i have modified for connection between UART-USB-adapter and BLE-module

communication works, but without a firmware(-dump) i see no chance to find out the existence of AT commands, but i think, some communication from bms to ble-module must exist, for instant to change the ble name and yes, the name is stored in ble adapter and can be change via bms

i will try to modify the existing software for other gadgets under https://github.com/pvvx/ATC_MiThermometer, wich offered encryption and change pin and name of ble adapter, but until now i haven't not successfully changed the parameter of ble connection, connections from pc to ble adapter via webinterface are ok, but from smartphone i can see the adapter,  but cant connect with ble terminals, only with webinterface