smagicld/xiaoxiangBMS #32: Balancing only when charging doesn’t work right.

thanarious, 2019-01-27:
Or the other way around. If I have “balance only when charging” enabled, everything works fine. However, if I disable it a keep “balancing g enables” active (of course), then the balancing works ONLY when NOT charging.

smagicld, 2019-01-27:
read the manual of your BMS to understand how it works.
The App is only UI to set the config params.

closed 2019-01-27 by smagicld

thanarious, 2019-01-27:
So you say it’s possible that it’s supposed to work that way? Seems illogical.

Anyway, there’s no manual, but there is a bundled android app which I will test in a while and report back.

thanarious, 2019-01-28:
I would prefer it if it would balance continuously, charging or not. Why would you want to balance only when NOT charging? You’ll lose at least 10h of balance.

smagicld, 2019-01-28:
Where did you get these 10h from? I'm charging my eBike batteries with 0.5C within 3h. So as you can see, it all depends on the use case. If your use case is to do proper balancing while charging, then you need an BMS with Balancer which can balance the charging current over single cels. Here is one that can do this: https://123electric.eu/products/123smartbms/

thanarious, 2019-01-28:
I use this BMS on a small 7s 0.8kWh solar bank made of junk cells from which I then charge my ebike through a dc-dc converter, as well as use for other low voltage stuff.

The solar bank is permanently connected to a charger. The way it works now, it gets around 3-4h of balancing while the charger is float-charging in CV mode. Looks like it keeps the cell groups in sync, so far.

However, if it was able to sync 24h, it would be better for when the cell groups drift a bit due to heavy usage.

I just can’t think of a use case where the user would need to balance ONLY when NOT charging and STOP balancing when charging starts. It just doesn’t make sense, for me at least...

desultorynz, 2019-02-04:
He's correct, why not make it so you can BALANCE during Charging and during not-charging (balance at both / all times) if cells are above xx voltage and more than xxx difference between the min & max cells.  Simply only being able to balance at charging (or at static idle) but not at both is crazy and undermines the real balancing ability of the BMS system.  What's the harm in being able to select "always smart balance" during charging and / or not-charging, just whenever it's needed.  This is a smarter approach.  Can this be added via the app you think to enable this?