smagicld/xiaoxiangBMS #33: Smart Active Balancing

desultorynz, 2019-02-04:
Currently in the iOS app it allows you to set the BMS for static balancing (balance cells only when the BMS detects NO charging and / or NO discharging), or you can set "Balance only while charging" which simply does just that and ONLY performs some (minor) balancing during charging.  However from using many batteries with your bms system now this method also fails quite badly if the charge current is higher than the low balancing current.  Eg if you are charging at 4Amps (or higher rate) but can only balance at 50-60mA rate then you can still end up with SOME cells showing "Over Voltage" and the pack out of balance.  Once over-voltage on any 1 cell occurs the BMS auto DISCONNECTS charging, however there is then NO balancing occurring and the rest of the cells in the pack still don't get balanced (or correctly fully charged).

I've looked at how many other similar BMS systems get around this issue and they use a method often referred to as "SMART ACTIVE BALANCING" which basically lets the BMS start balancing no matter whether it's charging, discharging or idle IF the cells are more than xx amount out of balance AND if they are above xx voltage (per cell).  If this SMART BALANCING was able to be enabled it would greatly enhance the real use usefulness of this Smart Bluetooth BMS systems as it would mean that BALANCING would occur more often and it would occur whenever it's needed (no matter what the battery is doing), and in turn if ever there was a cell that was showing over voltage (which then disconnects the charging) at least the pack could then go into AUTO BALANCE mode to help balance the pack so that if charging was restarted a few hours later then it would have further additional success.

In summary I shouldn't have to choose between using 1 or the other of the current balancing selections, why not make it so I can enable Smart Balancing and have it auto balance once cells are above xxx voltage and more than xxx amount out of balance.  By doing this it would really help the usefulness of your BMS system.

Please talk with your tech people to see how you may be able to implement this ability thanks.

thanarious, 2019-02-05:
As the developer of the iOS app has already mentioned, the iOS app is just a GUI for the BMS's build-in capabilities. So, if that's the way the BMS works, there is not much to be done, unfortunately.

desultorynz, 2019-02-05:
In that case the BMS internal software (for how it operates and leaves out of the factory) really needs an update so that the BMS supports balancing whenever it's needed no matter whether you are charging, discharging or whether the battery is sitting idle, etc if the cells are out of balance more than xx amount and the voltage is over xx volts.  This is how smart balancing works on many other bms units and with great success.

smagicld, 2019-06-10:
not related to the iOS App. can be done only on BMS firmware level.

closed 2019-06-10 by smagicld