smagicld/xiaoxiangBMS #34: Unable to read EEPROM

fsghzh, 2019-02-07:
Hi,

I run the latest testflight version of the APP on a JBD-SP20S007. On the config tab it is unable to read the current values of the EEPROM. It just shows me the 1 / 2 number and finally I get an error message "BMS response timeout". The standard APP from the App store is working and can read the values.

By

smagicld, 2019-02-07:
Hi

This was a broken Build. Please re-test the new build 2196 just released by apple (TestFlight)

BreitiDE, 2019-02-07:
hi, would it be possible to be invited to the testflight betas ?

smagicld, 2019-02-07:
yes, just send me your email to lukas at deszynski dot com

desultorynz, 2019-02-08:
Thanks Lukas, I just emailed you about an invite also, thanks and keep up the great work, cheers :)

smagicld, 2019-05-08:
fixed

closed 2019-05-08 by smagicld