smagicld/xiaoxiangBMS #36: Battery level notification request

temoest, 2019-03-10:
Hi
This is a great app, everyone with this type of battery should get it.

My feature request is to add an audio notification when the battery reaches 90% charge so ithat I can turn off the charger and extend the life of the battery by avoiding float at 100% charge.

Even better would be an option to switch the battery charging off automatically at a desired charge level. This would mimic the function of an expensive charger.

Hope this helps.

David

thanarious, 2019-03-12:
Disconnecting the battery (as per your "even better" request), would possibly damage an MPPT charger. The notification is a good idea, it could be implemented as soon as the logging does (hopefully through some integration with iCloud).

MPPT chargers are actually buck converters that use high frequencies and a number of capacitors to control output power to the battery. Disconnecting a battery when the charger is working in full could potentially raise the voltage on the capacitor above their limit, momentarily, that could possibly be enough to pop them.

temoest, 2019-03-12:
Thanks for letting me know about the potential damage by turning off the battery while it’s charging. The logging function is a great idea, I can’t wait for it.

smagicld, 2019-06-10:
Hi David

It is possible to implement it, but to have it as "protection" will not work.
The reason is: iOS can any time kill the BMS App. then you'll never get any notification.

But I'm working on general cell[v] and pack[v,soc] notification system (for information only). then you can set the rule to be notified.
for the time I'm closing the Issue.

closed 2019-06-10 by smagicld