smagicld/xiaoxiangBMS #38: Connects but displays nothing

dzubot, 2019-03-28:
Just bought the app in hopes I could use iPhone with my new battery but I cannot see any data on the app. First off I do not know if it has a xiaoxiang bms. I got a custom battery built in china for an electric paramotor. It came with a bluetooth dongle, I cannot open the battery up to see what bms it is as the battery is sealed. Can you tell me if it will work with your app or not?
[image]
[image]
[image]

apetre83, 2019-04-28:
I have the exact same issue. Connects but nothing is displays. And for me the bms name is actually xiaoxiang

smagicld, 2019-04-28:
Can you please add pictures of the BMS board and the link where you have ordered it?

Without these informations it is hard to investigate this issue. 

Thanks!

apetre83, 2019-04-28:
https://s.click.aliexpress.com/e/Kb3fzo4

smagicld, 2019-04-28:
Ok. It looks like the Bluetooth Dongle they sell is the old one from 2017. this is not supported (all my BMSs came with the newer one). 
Is it possible for you to send me the bms board + Bluetooth Dongle for reverse engineering? Only then I can implement support for it.

apetre83, 2019-04-28:
Its already on a ebike :) so cant really, but i can order some new
bluetooth modules and see if it works ...

smagicld, 2019-04-28:
Ok. I have just ordered the BMS from the link you sent to me. I should habe it in ca. 4 weeks and will need probably 1 Week to do some rev. eng to make it working.

thanarious, 2019-04-30:
> Ok. I have just ordered the BMS from the link you sent to me. I should habe it in ca. 4 weeks and will need probably 1 Week to do some rev. eng to make it working.

You're awesome! Thanx for doing this.

smagicld, 2019-05-15:
Guys, I have bad news. Today I have received the BMS from the same Ali seller. Indeed it has different Bluetooth module as from my seller.

[image]

But after connecting to BMS to test battery pack, I could see the BMS as Xiaoxiang BMS and I was able to connect to it and receive the data:
[image]

(do not look at cell voltage, I connected only 3 wires).

[image]

smagicld, 2019-05-15:
so it look for me like the connection between BMS Board and the Bluetooth Board is not working. Can it be that you have changed something on the wiring? (swapped TxD and RxD?) or the Pin on the connector has no contact?
did you connected the B- to the battery? connecting only the big connector (cell voltage sense) will not work, BMS will not have enough power to run.

apetre83, 2019-05-15:
Stage thing is that is working with the android app that they sent... its
partially in Chinese but works quite ok... we needed this app for ios and
it seem its not working. So nothing is wrong with the bms or bluetooth as
the android they send works fine

apetre83, 2019-05-15:
And againwht i see its the demo
Bms is not your actuall bms... tou are connecting to the demo bms that is
build in the app as this data was showing for my for the demo one.

smagicld, 2019-05-15:
No, the screenshot is of real BMS not the demo.

If you really want to have it working then you have to send me the BMS, together with the bluetooth dongle. I just spent 60 USD to test it myself and it works. I will not spent more money to buy another one.

After implementation I will sent it back to you. let me know if this is an option. If not, you have to live with the Android App.

apetre83, 2019-05-15:
Its fine man. This is the best you can do.

thanarious, 2019-05-16:
Lukas,
How come you're using protected cells on an e-bike battery? Never seen this before!

smagicld, 2019-05-16:
No, this is just my test / development pack - that's why I use protected - just in case.

For E-Bike I use unprotected LG's HG2. Here my refreshed Bosch Battery Pack with 666 Wh:
[image]

[image]

thanarious, 2019-05-17:
666Wh? Looks like the Dark Force is strong in this one!

closed 2019-06-10 by smagicld