smagicld/xiaoxiangBMS #39: “Remaining Capacity” meter question

thanarious, 2019-04-03:
Hi,
I know this doesn’t have anything to do with this app, it’s a BMS thing, but I would like your insights on the matter:
I have built a 7s low-quality cells battery. All cells have been tested and under 500mA they sum up to 16Ah. That’s the value I have set as “design capacity” through the app.
However, sometimes I see maximum voltage and 50% capacity (which would imply that the design capacity is much less) and sometimes I see 0% capacity with more that half the voltage (which would imply that the design capacity is much more).
What I have noticed is that the BMS does not detect low currents very good. Currents under 0.3A don’t get at all detected. If I calibrate it for very low currents then it does, but drifts a lot at high currents. Could this be the culprit?
Is there any way that the available capacity could be displayed in some other way, i.e. judging by the voltage?

thanarious, 2019-05-03:
An idea would be to have an option to display % judging by the voltage. There could also be a delay or even an extra "trainable" variable to adjust for voltage sag on high currents.

Here's an example:
Say I have a battery with Vmin: 21V and Vmax: 28.7V. Under no load, the % would be calculated according to the voltage, i.e. it would be 52% at a voltage of 25V. Under 10A load it would still should be at 52%, but the voltage would probably sag to under 23V in a high-ir battery. So the application could be "trained" to adjust the % depending on the current and would further adjust the correct % when there is no current going through.

Please let me know if this is by any chance useful and if you would need any real-world data to do some number-crunching.

Keep up the good work.

smagicld, 2019-05-08:
First thing: the BMS has very unprecise current measurement. It is not linear too. It means if you calibrate it for low current (say 500mA), then it does not match for 15 Amps discharge. Because I use the BMS's mostly for building eBikes batteries, I calibrate them with 5 Amps discharge.
If you need better values, you have to go fro BMS hardware > 500 USD. These China BMS are not build for precision.

It is also very hard to determine the remaining capacity based only on voltage. As you already wrote, the Call / Pack voltage depends on the drawn current. Then there are other parameters like type of battery, discharge line, age, env. temperature and many others parameters.

I do not know what is your use case, but maybe you could add dedicated Capacity meter. For my big battery bank (20kWh Home Battery) I use Victron BMV 602S [Shop link](https://www.photovoltaik-shop.com/batteriemonitor-victron-bmv-602s-black.html).

This meter is connected to raspberry PI with an RS232 <-> LAN adapter [Connection example](http://www.deszynski.com/wp/victron-bmv-700-lan-port/) so I can display everything on an WebSite: [My Solar](https://solar.deszynski.com/solar/)

Maybe this solution can help you.

thanarious, 2019-05-09:
Here's my experimental setup, getting more hardware to do a 5kWh build soon. I don't think I will be going the Batrium way. Is there a DIY lithium BMS/balancing solution from Victron? I am already using their solar chargers h/w, ramping up for a raspberry running Venus OS, but never seen a BMS from them that would manage a DIY lithium bank, as Batrium does.

aaardvaarkau, 2020-08-17:
I notice the same issue with current calibration and the strange results for State Of Charge (= Remaining Capacity).  My current measurements were all linear from 1 to 10 amp but: a) a lot of noise and variability, and b) same as you said, nothing under 0.3A).

I really can’t work out how the BMS is trying to calculate SOC but notice the following (I am using this app and the BMS manufacturer’s (LLT) own Android app, I don’t have the PC software):

1. Tonight when my battery was nearly fully charged the apps reported 10%, then when I did a small discharge followed by a small charge, it suddenly changed to 80%.

2. The brief information from LLT about measuring SOC/capacity is hard to understand, but suggests that they do coulomb counting (ie. measure current over time to work out AHs in and out), but only do this  sometimes. At other times they seem to use cell voltage. Their app allows you to set battery voltages at different % capacity, so they must use that for something. 

3. Using battery voltage is almost impossible because most lithiums have a VERY flat discharge curve. Mine for example is completely flat on discharge from nearly full to 70% - no voltage change at all. As you point out temperature and charge/discharge current will make more difference than state of charge.

4. It would be really useful to know how the BMS tries to calculate SOC. I have never got any answer from LLT to my emails, does anyone have a means of communication with them?

5. I purchased this app because I was hoping it would measure SOC better than LLT app, so am disappointed to find that it seems to be a hardware problem.  On the other hand the LLT “user notes” imply that there is a sophisticated coulomb counting system in there somewhere, maybe we just have to understand how to make it work.  (But it *was* worth buying this app just for the voltage calibration function, so thanks for that).

smagicld, 2020-08-17:
Plaste Note this is very cheap China stuff. It uses simple resistor for current measurements. The current measures by this bms are not linear, for example: if I calibrate the bms with load 10A to 10A, then if I reduce the load to 2A the bms shows me 2.2A (compared to my fluke multimeter). 

So to get more useful results you should do the calibration with the load you are using the BMS. 

 if you really want precise SOC measurements - you have to go for column meter like victron BMV-700 - but it measures only the capacity and costs 7 times the BMS.

aaardvaarkau, 2020-08-18:
Yes it is quite cheap (though my 120A unit cost me nearly AUD100) but the problem may not be that it uses a resistor for current measurement (so does an expensive DVM), but that the resistance is so low that the voltage across gets noisy as current falls.  The resistance has to be small so that you don’t lose much voltage across the BMS. I measure total resistance across the unit at 0.4 mΩ, which includes the FETs.  If we say half is FET then the current monitor resistor is 0.2mΩ.  With  300mA current this is 70microvolts.  If the A/D converter in this chip can measure to about 0.1mV resolution/stablility (reasonable guess for 14-bit ADC measuring 1V full scale) then the minimum readable current = V/R = 0.1mV/0.2mΩ = 0.5A which is about what we see.  I guess better hardware might use switchable shunt resistors, or better ADCs.

 

I am checking accuracy of the SOC reading today.  So far it looks good on a straight discharge at 10A.  The manufacturer Xiaoxiang/LLT says you have to run a complete charge / discharge / charge cycle so that the SOC calculation can ‘learn’ from your discharge and charge curve.  So starting at full, I’m discharging to 2.9V / cell which will register as empty (because I have set “Empty” voltage to 3.0V).  Then will recharge it and see what happens.  With 10A discharge, so far at 50% it is extremely accurate.  But I suspect if I charge or discharge at less than 0.4A the accuracy will be very poor, so will check that next. Also need to check how it works on charge, and when alternating between charge and discharge.

 

I have a Battery Monitor like the Victron  (a NASA BM1) which has been OK with my AGM battery.  But I assume it won’t be so good with a lithium battery because it’s parameters all assume lead acid.  That is why I bought this app! Because I was hoping it would do the SOC instead of the BM1.  Will have to see how both work in practice, so a few more tests to run.