smagicld/xiaoxiangBMS #43: Feature request: battery cell database with recommended setting

BreitiDE, 2019-05-19:
hi,
it would be great to have a cell database of batterys with possible chance to take over a recommendation to config. this can first start with the possibility to add cell by your own and may extend to a common database (csv or whatever not too complex 👍. in real youthan can select the cell type, how many in serial and add a secure level of percentage which kan be 0% for hardlimmited settings and maybe 10% for soft ans secure settings ... this can may save beginners for overtewaking ... what do you guys think ?

smagicld, 2019-05-19:
Hi
In my opinion, someone who is playing with BMS cell / configs should exactly know all this stuff. It is nothing for beginners. wrong settings here can end with battery pack in fire.

But I the people are asking for it... here my proposal:
currently I'm implementing settings save and load from files.

We could use it to share battery packs settings! On GIT we could have directory / page with recommended settings for specific S (serial), P (parallel) and cell type (for example LG H2).
Additionally we could share details how to build the battery pack (I can give infos on my own Bosch e-Bike extensions).

So if you like my proposal, we can do it. All I need is someone to maintain this GIT subpage. any freelancers?

thanarious, 2019-05-20:
And while you're at it, especially now that setting the under/over voltage parameters is available in "Protections" part of the Settings, (it's been working as expected on the TestFlight version btw),  maybe it would be a good idea to add a "disclaimer" text somewhere in the app, informing users that -indeed- this is not a toy, nor a "guide", but a tool; and that it should be used by people who know what they're doing. Just as you did on the Calibration menus. Altering protection parameters could be even more important, tbh, so maybe they deserve a menu be themselves, that would probably need a GUI redesign of the "config" menu.

BreitiDE, 2019-05-20:
ok, I am with you guys ... so let me change my idea to ... would it be great to save the used cell type to an device and link to an internal/external database where I can get the cell specific information ? :) But as more I think about the request ... it might be more nice to have and convenient...