smagicld/xiaoxiangBMS #44: Combine discharge value with time

victornorberg, 2019-05-23:
Hello.
Thank you for a great app!

Under "discharge", why not combine the amps, voltage and a clock to get a value in wh and or kwh? Energy used multiplied by time.

That way you can track how much energy you are using on a certain trip. 

Victor , E-moped builder in Sweden.

victornorberg, 2019-05-23:
and/or implement "trips" section. To start and stop measurement of this data.

thanarious, 2019-05-23:
That's a neat idea; you could even use this to calculate the actual capacity of the battery. Fully charge, zero data and discharge!

victornorberg, 2019-05-23:
It gives you a ballpark at least.

I think the android version of the app does this.

martinbradshaw, 2019-05-23:
this is the feature i would like most to see

smagicld, 2019-06-10:
Victor, if you want it, I can make it. 

Just take a piece of paper and draw sample screen (mockup) how you would like to have it implemented.

I myself drive a lot with this BMS on my e-Bike and do not need of it. But if there are more ppl who are asking for it, why not....

victornorberg, 2019-06-10:
Nice!

Would it be hard to add a separate page for "drive mode" where it looks more like the android version? I could make it, all the visuals if you want.

If not the above then...

Simplest version would be to include this data where the other data is today, at max discharge charge value. People that are running regen would like to see how many Ah they regen during their rides. That would also be possible to add.

If you dont have time to do it, maybe I can do it. But I am very green in this field.

Love your work!

Victor

CSYGD, 2020-09-21:
Hi!
I am uping this conv. Victor did you send some Mockup about a new trip section? Did I miss something over the past year? I do not have Android only iOS. I am also very interested by this function Lukas.