smagicld/xiaoxiangBMS #45: auto shutoff

martinbradshaw, 2019-05-23:
Would be cool if the charge and discharge would turn off when it detected charging at 100 percent, think it wouild be safer to leave.

thanarious, 2019-05-24:
This would be a mistake. The charger wouldn't like it if the BMS disconnected the battery all of a sudden, this could possibly kill the charger.

martinbradshaw, 2019-05-24:
the charger gets disconnected all of a sudden when i unplug it, whats difference, the chargers tail off the current near the end so it should be okay.

smagicld, 2019-05-24:
This kind of functionality should be never implemented on App side. Imagine iOS will kill the app for any reason, what then?
Anyhow the BMS board is doing this already (this is the reason why we have the BMS) - as soon the pack voltage or single cell voltage reaches the maximum - the mosfet gets off (you can see it on the over voltage counters). And even better - if the voltage gets below threshold (for example 99%) - it will charge the battery automatically so you have always full battery when you need it.

closed 2019-05-24 by smagicld

thanarious, 2019-05-30:
> the charger gets disconnected all of a sudden when i unplug it, whats difference, the chargers tail off the current near the end so it should be okay.

You should really disconnect the charger from the wall first, and then disconnect the battery from the charger.