smagicld/xiaoxiangBMS #47: Discharge Value

mersebikes, 2019-06-04:
The newest update doesn’t allow setting current values correctly on 6 figure capable boards 

By default mine loads as 120000 ‘mV’ this should be updated to mA I guess 

And if I modify I can at maximum put 5 digits so can not go above 99999 for example 

Thanks for a speedy fix this app saves me having to carry an Android phone around now the setting current has been added!

smagicld, 2019-06-04:
Thanks for your feedback. 
As you wrote, it should be mA instead of mV. 
I‘ll extend these 2 fields to 6 digits and submit new release tomorrow.

smagicld, 2019-06-05:
Just FYI: submitted to Apple. Should be released in 1 - 2 days.

[image]

closed 2019-06-05 by smagicld