smagicld/xiaoxiangBMS #51: S0005

DnzDev01, 2019-07-27:
I hope the app can do landscape view
And hav a chart live feed
And alao a notification for differential voltage

smagicld, 2019-07-27:
Hi. 
What do you mean with "And alao a notification for differential voltage"? Delta between Cell Voltages?
What do you mean with "And hav a chart live feed"? have no idea what you mean.

Please try to describe your requests in for of an user story in following format:
As a < type of user >, I want < some goal > so that < some reason >.

Additionally you could bring some real examples.

The landscape stuff I'll check today.

DnzDev01, 2019-07-27:
Delta between cell voltages sir
So it will notify me if its above my preset value limit

DnzDev01, 2019-07-27:
An energy chart history

Where we can browse the charge harvested history and the discharge accumulated history 
Showed by bars, organize by day

DnzDev01, 2019-07-27:
Thank u sir i hope u make d app more powerful 👍🏻😇

This will be used by many

DnzDev01, 2019-07-30:
Sir Thank you very much for new update!! Appreciate it! 

I hope next is the history bar graph of both discharged and charged (organize by day)

Thanks!!

closed 2019-09-06 by DnzDev01