smagicld/xiaoxiangBMS #52: Android password

Wispje, 2019-08-03:
Just changed the password in the android app and now reading data doesn't work in your iOS app. Is it possible that you need the generic code of 123456 to have access?

If this is the case, it would be useful the program a field where I can entre that password in the iOS app because I don't want it to be the generic code.

smagicld, 2019-08-03:
Sorry, but i can not help you. I have no android device to reproduce your issue.

Wispje, 2019-08-05:
I'll change it back and see if that fixes the issue!

Wispje, 2019-08-08:
Changed the bms back to the generic password but still not able to read BMS. So it's probably an unrelated issue. I get the following:
Error
No response from BMS. Try to read/write again

Before I never had this issue

smagicld, 2019-08-09:
Switch off the password in the Android app.

Wispje, 2019-08-09:
Mmm, It seems like you cannot switch if off.

smagicld, 2019-08-09:
Sorry, at the moment I can not reproduce it because I do not have any android device. Maybe I’ll get one next month to look on this.

smagicld, 2019-08-11:
Hi. I've got an Android device to reproduce your problem. But I can not find any place to set BMS Password (only the App password).

Can you paste me some screenshots how you have set the pin on android?

Thanks.

This are all the settings I see with my BMS on Android App:
[image]

Wispje, 2019-08-11:
Hey, the password is indeed only in the app setting tab. That's why my issue is most likely not related to the password change

Wispje, 2019-08-11:
Just dit some testing. A friend that also has the app on his phone cannot load parameters as well. But with android I can still load parameters and make changes. Very strange issue

Wispje, 2019-08-11:
I just was able to load parameters! I have the impression when I make some changes with android and then quickly connect with iPhone that I can load parameters. Then, after a short time, I'm not able to load parameters no more on iPhone.

smagicld, 2019-08-11:
but you know, if you connect to the BMS with Android or iOS then the BMS is during the connection time not visible over Bluetooth? Probably this is your case.

If you do not provide me the step by step guide with screenshots to reproduce your issue, then I will close it, ok?

Wispje, 2019-08-11:
Yeah, I know it can only be connected to one device at a time. this is not my issue.

You can close it since it has nothing to do with the android password. I just have bad connection somehow.

smagicld, 2019-08-12:
issue not related to the app. closing.

closed 2019-08-12 by smagicld