smagicld/xiaoxiangBMS #58: App crashes

gixxer001, 2019-11-12:
I can connect to the bms with my iOS devices, but when I try the in app purchase so that I can change the settings it crashes.

smagicld, 2019-11-12:
Can you provide me more details please?
What is the app version? (1.2.7?). What is the iOS version? How can I reproduce the problem?

gixxer001, 2019-11-12:
iOS version 13.2.2 and app version 2.0.0
I downloaded the app today as I just got my custom battery. In order to turn the battery on/off or configure any settings I have to upgrade to the “pro” version of the app for 5.99. When I go to in-app purchases to attempt the transaction the app crashes.

smagicld, 2019-11-12:
I've tried to reproduce your problem - it works on all my devices - i can do the purchase.
The Version 2.0.0 is not yet released, so it is impossible you have it.

Pleas tell me the version and build number (within the Logo)

gixxer001, 2019-11-12:
My mistake the version and build is 1.2.8, build 4515.

smagicld, 2019-11-12:
Probably I've found the problem - it is related to the google fabric framework. the version I'm using is causing problems with iOS 13. I've prepared new version (1.2.9), should be available within next 2 days in AppStore

gixxer001, 2019-11-13:
Excellent. I’ll keep an eye out for the update.

gixxer001, 2019-11-13:
I updated the app and it still crashes during the attempted in app purchase.

smagicld, 2019-11-13:
thanks for your feedback. I'm waiting now for Apple's crash report, hope to get one.

did you tried already to remove the app and install it again (from AppStore)?

gixxer001, 2019-11-13:
Yes. I deleted and reinstalled. I hit info, in app purchases, in hangs in the next screen saying fetching, then crashes.

smagicld, 2019-11-13:
please drop me an e-mail to lukas at deszynski dot com. I'll build test version and give it to you for testing.

gixxer001, 2019-11-13:
I sent an email to Lukas@deszynski.com. Thanks

smagicld, 2019-11-15:
fixed with 1.2.10

closed 2019-11-15 by smagicld