smagicld/xiaoxiangBMS #59: Restore Purchases after App changed to free

mersebikes, 2019-11-15:
I used this app on multiple devices in my household using the iOS family sharing 

My brother recently got a new phone and I can not restore the purchase since now the app changed to free and the on/off feature is locked with the 5.99 pro version 

I have the pro version on my phone still but via the initial route back when the app was paid for 

Is there any way around this or need to rebuy the app and if I change my phone will I also lose the app and need to pay again

smagicld, 2019-11-15:
Hi
The app has changed to free months ago. there were two month for everyone to get the pro feature for free. If you have bought the Pro version for 0.00 at this time, then simply press Restore Purchases.

All purchases are linked to appleId account, so even if you change your phone - you still have it.

Note: In-App purchases do not work with Family sharing (because every family member has its own appleId)

mersebikes, 2019-11-15:
Hello 

Thanks for the quick response I don’t know if it’s maybe an Apple issue then 

It was working fine until this recent phone upgrade so never noticed any change  

When I click restore purchase the app closes and when reopened it asks again until eventually it says no purchase found 

Tried a couple times, is it still iOS family share compatible or can that be disabled 

Just helps to not have to give my brother my phone all the time maybe he will have to buy it

smagicld, 2019-11-15:
yes, there was a bug with iOS 13 which is causing the in-App screen to crash on some devices. 20 Minutes ago I've released the 1.2.10 Version which is fixing it. check your app store for the update and once you have 1.2.10 - do restore again.

smagicld, 2020-01-02:
fixed with 1.2.10

closed 2020-01-02 by smagicld