smagicld/xiaoxiangBMS #61: Notification

DnzDev01, 2019-12-16:
About notification 
when notifying me about “Cell delta XXmV is greater than XXmV”

I hope it point out, which CELL its referring to.
Sample:
“Cell 8 delta XXmV less than XXmV”

Thanks

smagicld, 2019-12-30:
Hi

Cell Delta is always between cell Min Voltage and cell Max Voltage, so you can not reference one cell.

I could extend the message like this:
Cell delta **XX** mV is greater than defined **XX** mV. CellMin[**1**] **3.123**V, CellMax[**5**] **3.170**V.

Number in brackets [X] is the cell number.

let me know if this can help you.

DnzDev01, 2019-12-31:
Yes sir that would be helpful. It would be also nice if the notification also provide that information. 

Like when i set my notification on “cell delta is greater than 50mV “ warn me. [sample pic]

Tnx very helpful to note me which cell is causing difference
[image]

smagicld, 2020-01-02:
[image]

so this would be the final notification. ok?

DnzDev01, 2020-01-02:
Yes sir!! Perfect! 

Pls include it also in the App live interface. Tnx. 

Will wait for the update

closed 2020-01-02 by DnzDev01