smagicld/xiaoxiangBMS #63: BMS not sending any data.

stanislas8, 2020-01-21:
Hello, I have an issue on my BMS where I don't receive any data on it. 

It has been like this for a few month (not a recent issue) I was doing some setting modifications on it, (basic battery reading and writing settings), saving them, and when I restarted the BMS, it just won't read any data coming from it anymore.
With my Iphone, I can still detect the BMS, but not with an Android device. I can still connect it (as on the pic below) but nothing is shown. I've tried to delete+download the app again, tried it on other Ios devices, nothing worked.
When I check my bluetooth module from my BMS, when it's trying to connect to it on my Iphone (on the screen shown below), the indication light does light up.

I don't know what's wrong and at the moment my only option is to buy another one :/....

Thanks for the help
Stan

[image]

thanarious, 2020-03-04:
First try disconnecting the bluetooth dongle for a couple of minutes. In that doesn't work, disconnecting the BMS from the battery should do the trick. If it doesn't, probably the BMS' 5V power supply has died and the BMS is not working either. Do you get output from the BMS under load?

dt1400, 2020-03-06:
Delete the app and reinstall and try again. It worked for me. I did a low voltage balance test and set. Balanced with my battery checker, recharged and reconnected and it would not engage. After several different try’s, I deleted and app and reinstalled it and everything came back. Hey... your not doing anything anyway, try it, what do you have to lose?

rfndzc, 2020-04-27:
I have a similar issue except that I am able to connect to my module via Android app, and it populated the data fine.  The iOs app does not populate data.  I have tried resetting.  It may be too low voltage on the first cell if that is the one that powers the device...

rfndzc, 2020-04-27:
I have a similar issue except that I am able to connect to my module via Android app, and it populated the data fine. The iOs app does not populate data. I have tried delete the app and reinstalling.  Any update on this?

smagicld, 2020-04-27:
I think the problem is not related with the app. I suspect some firmware version which is not supported. without having the HW it is impossible to fix it.

Question: would it be possible to get this BMS board for investigation? I would send it back to you after I fix it.
PS: I'm not willing to buy same model because I've done it in the past (two times) - and the delivered devices were working without any problems

rfndzc, 2020-04-27:
Where are you located?  I would much prefer to work with you on it if at
all possible, in the hopes that would be faster, I would not have to remove
the bms from the installation and I might also learn something.  I captured
the advertised UUIDs:  FFE0 and FEE7.  FF00 did not appear.

One other thing worth noting is that in the bluetooth settings when I try
to connect the device is ‘unsupported’.  I am not prompted to enter the
1234 pin, as I am on Android when it connects successfully.  Is there any
other information I can provide?  Thank you

smagicld, 2020-04-27:
Switzerland.

There is one option how you could help me:
1. connect to pc/mac USB -> rs232 dongle (important TTL level) to your pc.
2. connect dongle RX (receive) to BMS TxD and RxD over an 1n4148 (or similar) diode
3. start software for capturing rs232 data (binary mode)
4. start capturing and start the android app
5. send me the details

below details how the connection should look like. 

let me know if you can do it. if yes, I'll provide you more details to the SW (for RS port capturing)

[image]

rfndzc, 2020-04-28:
I am showing an anomalous voltage at cell number 1 on the Android app, fluctuating around ~0.080v.  Was not able to fix it while still installed so I decided to pull the BMS out entirely.  Is this normal behavior at cell number one?  I checked the wire lead and it is fine, and the cell itself shows the correct voltage when I measure that (3.2v).  I will take it apart again tomorrow to clean and look at the bluetooth module to see if I can be of help.  Any thoughts on the cell no. 1 voltage issue?  Thanks

smagicld, 2020-04-28:
from my experience: I had once floating voltages and instabile Bluetooth connections when I connected the BMS to battery pack WITHOUT connecting B- to the battery. Just the voltage sensing wires are not enough. is your B- connected?

stanislas8, 2020-06-25:
I finally found the issue and solved my problem. 
For some reason, while manipulating my ios device, my device name got erased. There was no option for me to go back and connect to the BMS again. (This happened in December 2019)

I was messing around with my android phone today and installed a new version of the LTT BMS app, I saw the device name blank in the settings. 
So I entered something and tried to connect it again using my Ios device, now everything works back again. 

Before this, I tried to connect many Ios devices without any success. Perhaps, when the device name goes missing, the Ios app won't read the data?

smagicld, 2020-06-26:
Do you remember the name you of bms you had before? Or was it just blank string?
I would like to reproduce it. Can not imagine it has something to do with not receiving data because the name is used only to display the name - it has no other functionality. Internally the communication happens over bms UUID.

stanislas8, 2020-06-26:
> Do you remember the name you of bms you had before? Or was it just blank string? I would like to reproduce it. Can not imagine it has something to do with not receiving data because the name is used [...]

[image]

I believe it was called like this. But when I went on my android phone, there wasn't anything so I entered "stan", and now the name shows up as "stan"

planetxxx, 2020-06-27:
The same thing happen to me after I changed settings, it was a brand new bms just got, changed my settings, saved configuration, bms write, then I disconnected then I tried to get back in to make sure everything was ok, loaded app saw bms connected but would not load got the same screen as he pictured. So I disconnected the power to the bms unplugged dongle then I powered the bms plugged in dongle then all was good again.

smagicld, 2020-09-18:
fixed with 1.2.18

closed 2020-09-18 by smagicld