smagicld/xiaoxiangBMS #68: Won’t reconnect

dt1400, 2020-03-03:
I was doing some programming and disconnected the bms to double check my results with a battery checker.... and it confirmed what the bms said. So I reconnected the bms 4s with temperature protection. The blue light blinked when I reconnected and did not stay on. I cannot reconnect with phone app. It doesn’t show up on my phone ( blue tooth) to connect to anymore. It’s a head scratcher. How do I remedy this? Ty for any ideas. dave

smagicld, 2020-03-04:
I would try this:
1. Disconnect bms from battery pack (all cables (Bluetooth dongle too!)). Wait 1min an connect again battery pack again BUT without BT dongle
2. Connect the BMS to the PC software over provided usb-rs232 interface. 
3. If you can read data over PC, connect BT dongle. If PC readout does not work - contact your bms seller.

thanarious, 2020-03-04:
First try disconnecting the bluetooth dongle for a couple of minutes. In that doesn't work, disconnecting the BMS from the battery should do the trick. If it doesn't, probably the BMS' 5V power supply has died and the BMS is not working either. Do you get output from the BMS under load?

dt1400, 2020-03-04:
When I hook up solar or a work bench power supply it does no go through the bms. It does not register.

dt1400, 2020-03-04:
Should I use my wife pc or could I use my Apple laptop?

dt1400, 2020-03-04:
Ok I looked jab-rs232. It would take a desk top pc. We do not own a desk top, only lap tops.

thanarious, 2020-03-05:
> When I hook up solar or a work bench power supply it does no go through the bms. It does not register.

If it doesn't charge, then probably the BMS is bust. What if you try to pull some amps off of the battery?

dt1400, 2020-03-05:
The point is I cannot monitor the new Fortune batteries. 8 each 12v. So I have no idea what’s going on. I have bottom balanced them waiting for Batteryhookup.com to respond. It’s been a week. I have sent emails and phone calls. No response.