smagicld/xiaoxiangBMS #70: Low Temperature Protection Problem

AliciaV69, 2020-03-15:
Today I went below the low temperature lower limit. I saw the BMS charge for two seconds, disable charging, the the battery was providing power for about 2 seconds, then it went back to charging mode.  It repeated this cycle for over a thousand times. This behavior could potentially destroy my battery. My expectation was that once the low temperature limit was passed the charging port would be disabled until the upper temperature was reached. I have a video of the behavior if you are interested

smagicld, 2020-03-16:
Hi. I’m not manufacturer of this BMS, the app is just an UI. If you feel your bms is not working correctly, please contact your seller. 

Hint: check your protection settings. There are always two values - trigger and release. In your case the bms should switch back on when release temp is reached, not the higher temp.

thanarious, 2020-03-16:
Also, it’s better to cut the charging from the charge controller, because taking the battery offline under heavy current when charging, could very well damage the controller.

kylelongstaff, 2020-04-11:
Hey, I'm having a similar problem with BMS except its because I disconnected a couple of temperature sensors. 
Can you possibly make a function that disables temperature sensors? or so I can select only 1 to use?

smagicld, 2020-04-12:
I'll add in next release (next week?) option to setup number of NTC sensors (0 ... 8). it should be possible to disable them setting to 0.

kylelongstaff, 2020-04-13:
Wow thanks a ton! This means my bms can have temp protection again! ;)

smagicld, 2020-04-13:
The App is currently in review by apple. I expect they release it tomorrow.
The NTC setting you'll find at very end of the BMS settings screen:

[image]

kylelongstaff, 2020-04-13:
Perfect that's exactly what I need! thanks!

smagicld, 2020-04-15:
could you try it? does it work for you?

RCinFLA, 2022-02-05:
This relates to the post requesting 'charging' and 'discharging' on-off capability.

The only way to have a BMS reliably prevent low temp charging is to totally shut down BMS at cold temp.

Problem is disabling MOSFET responsible for passing charge current results in an additional diode voltage drop for discharging current.  This extra voltage drop causes a lot of self heating if discharge current is very high, typically greater than about 5 amps.

Most BMS's will override the low temp charging MOSFET shutdown if too much discharge current is detected.  It will re-disable the charging pass MOSFET when discharge current drops to a low level or reverse charging current is detected.

This is likely what you are seeing.

Most BMS's do not have strong series pass MOSFET gate driver.  The just use a gate pull up resistor to turn on the N-ch MOSFET's.  Since there are many large MOSFET's in parallel there is a lot of gate capacitance.  With low gate drive current it takes a few milliseconds to turn MOSFET's on and off.  Add on the time it takes for BMS to sense the pass current level and you have the oscillation you are seeing.  If you lower the discharge load current on BMS it will stop the oscillation.