smagicld/xiaoxiangBMS #72: Reversed and Wrong Charge/Discharge

AliciaV69, 2020-04-02:
My BMS is showing a discharge current while I’m charging and a charging current while discharging. The magnitude of the current is off by a factor of about 3.5. If I trigger the low temperature cutoff it does show the charging port being disconnected (in this case while discharging). Is this fixable by calibrating the current sensor?  How do you do that?  I’ve tried two different BMS boards (same model) and they both work the same)

smagicld, 2020-04-02:
It looks like you’ve wired your BMS wrongly. I suggest you to contact your BMS reseller with this problem as the App is just an UI and displays what the BMS is reporting.

AliciaV69, 2020-04-02:
It’s wired the way the seller told me with a older version of the board

Alicia Viskoe

AliciaV69, 2020-04-02:
And wrong wiring doesn’t explain the incorrect current values

Alicia Viskoe

On Apr 2, 2020, at 9:05 AM, Alicia Viskoe <alicia_viskoe@hotmail.com> wrote:

 It’s wired the way the seller told me with a older version of the board

Alicia Viskoe

smagicld, 2020-04-02:
it writes the calibration results to the BMS. first calibrate zero current, then connect load - and enter measured value. then connect charger and enter measured value.

closed 2020-12-20 by smagicld