smagicld/xiaoxiangBMS #75: Temperature sensor discrepancy

martinw123, 2020-04-15:
Temperature sensor 1 is showing the correct temperature but temperature sensor 2 is showing a fixed value of -30 degrees Celsius. I can’t find a way to disable temperature sensor 2.

smagicld, 2020-04-15:
With the latest version (1.2.13) go to bms settings. On the very bottom you’ll find the option to disable or enable NTC.

martinw123, 2020-04-15:
Hi,

Thanks for the reply. Only the box “NTC 1” is ticked but I’m still seeing 1 and 2 on the main screen. I couldn’t find any reference to “NTC” in the instructions so don’t know if it’s OK to leave it like that?

smagicld, 2020-04-15:
then re-write the settings. switch all NTC off -> write to bms. then enable just one -> write bms and check result. let me know.

martinw123, 2020-04-15:
Ticking Or unticking the NTC box makes no difference. I did as you said and both Temp. Sensor 1 and 2 are still shown (1 is correct, 2 is -30). I checked that writing to the BMS had worked by performing a “read” after unticking the NTC 1 box. The box remained unticked but it didn’t change anything else.

smagicld, 2020-04-15:
you can write to the BMS only if you perform changes. 
I can not imagine how you can see still two temps if you disable both NTC. I did tested it on different BMS's and it works for me.

Try to do the following:
1. Untick all NTC AND write to BMS. Create an screenshot!
2. Go back into main screen of the app (the very first screen where you can select BMS to connect)
3. Connect to your BMS
4. create screenshot
5. post here the screenshots

martinw123, 2020-04-15:
[image]
[image]

smagicld, 2020-04-15:
ok. and you have also disconnected form BMS after you've changed the setting? (or just kill the App and connect again).

martinw123, 2020-04-15:
Yes, I’ve tried that. The NTC box remains unticked but the temperature sensors are still present on the front page. Thank you for trying to help - I don’t know what else I can post to try to help.

smagicld, 2020-04-15:
the last thing you can try is to connect the BMS to Windows PC with the manufacturer USB-Serial interface / software to check it.
I have on my place 5 different BMSs and for all of them I can switch on/off NTC's.

have tried, but can not reproduce your problem.

martinw123, 2020-04-15:
Thanks for trying. I don’t have the usb to serial interface so I will have to leave it as it is. I appreciate your time.

smagicld, 2020-04-15:
Just to show you it should work. do you have another BMS board to test it? could be HW problem, but I still do not understand why your BMS is reporting two temps when you disable all NTCs.... really strange

[image]
[image]
[image]

martinw123, 2020-04-15:
Unfortunately I don’t have another BMS at the moment but I have ordered one from another manufacturer so I’ll test it when it arrives and let you know.

smagicld, 2020-04-15:
cool, and please let me know! I'll leave this thread open.

lazycat38, 2021-01-13:
Hi, I'm currently experiencing exact same issue. My bms came with two temperature sensor. Temp 1 sensor is working correctly but temp 2 sensor is fixed at -30c. I dont have the ability to write or changing bms parameter yet because I'm afraid even if I pay for it, I could not adjust it like @martinw123 . Please provide further instruction for this situation.
Thank you

smagicld, 2021-01-13:
It can be that the second temp sensor is broken. Have you tried to swap them?
The app itself is just displaying what the BMS is sending, so it is more likely HW problem.

martinw123, 2021-01-13:
Hi,

There is no second temperature sensor. I don’t know why the software keeps showing it abs does not allow the user to de-select it.

Thank you

smagicld, 2021-01-13:
Then it means it is enabled. Disable it in the settings. If you do not want to buy the pro - use the bms software for pc (ask your reseller) and the usb-> rs232 dongle.

martinw123, 2021-01-14:
Hi,

We have been through this before - disabling it in the settings doesn’t do anything. The temperature still shows on the front screen. 

I have the pro version and I just live with this software issue because it doesn’t have any impact on what I am doing.

smagicld, 2021-01-14:
If you think it is SW problem, we can investigate it. 

Just go into info screen and tap once on the version / build number (within logo). 
you should get an confirmation.

connect to the bms.  Wait for first data. Go back to info screen and tap again on version string. 

An e-Mail send Dialog will appear. Send me the debug data.