smagicld/xiaoxiangBMS #81: Configure of Discharge rate in %?

Rafale, 2020-06-15:
Hi, what does the discharge rate in % mean?
My battery’s discharge rate is 0.5C , how should I put it in %?

smagicld, 2020-06-15:
Hi Raffaele

C is the capacity. 
So you have to calculate the percentage from the total capacity into percentage.

Best regards
Lukas

Rafale, 2020-06-15:
So, my battery total capacity is 0.5C , which mean the % is 50% ? 
Thx for the fast response and support!

aaardvaarkau, 2020-08-17:
I think (guessing because I can’t find much info on this app) that this might refer to “Self-discharge Rate”, which (also guessing) might be the rate in mA at which the battery self discharged. Which would be low, perhaps 10 mA for my 120AH 4S lifepo4 battery.  Grateful for confirmation of this, but it might explain why some people are having trouble entering big numbers. Somewhere the bms needs to know this figure if it is to accurately calculate SOC.

smagicld, 2020-08-17:
Unfortunately it is not described in the BMS HW document. the only thing I found in the docu is: "Self-discharge rate: the number of self-discharge capacity for the battery pack, the general value will be relatively small."

But I can imagine why: the reason is, that the value strongly depends on the batteries you are using. you should look into thei datasheet and find information about self discharging level. than based on this, you cell configuration you can estimate the self discharging rate. 

I personally never do this, just keep the BMS default value.

aaardvaarkau, 2020-08-17:
That sounds right, but suggest you re-label in the settings from “discharge
rate” to “self-discharge rate”.  The BMS will use this to update and
calculate state of charge.  I wouldn’t measure it but would adjust it if my
SOC figures are wrong when battery is in storage for a while.  As you say,
use the default unless you feel the need to change it.

But the problem is, people think it is for the discharge rate they use in
operation eg 0.5C or 10 amps, so they are confused.

While you are there, is there any documentation on voltage and current
calibration?

JR

smagicld, 2020-08-17:
good point. I took the labels as in the PC App. I'll rename it to self discharge.

to calibration: I've made some pictures how I do the calibration, but had no time yet to put all together becuase of my regular job. Maybe during my September holidays....

aaardvaarkau, 2020-08-17:
To be clear, the self discharge rate is the rate at which your battery loses charge if left unconnected. It seems that the manufacturer wants it in %, not mA as I suggested. But that doesn’t answer the question “% over what period”.  Maybe it’s % per month but you’d have to get that from the chip manufacturer I guess. As agreed, users can best use the default figure as a start point and then adjust it if/when you leave the battery in storage for a few months and remember to measure remaining capacity before recharging it. Not so easy, but probably most people won’t need it anyway.

RaWolt, 2020-11-09:
My sugestions: The question is: For what they want to use this value here? 

1) The "Self-discharge" rate from the cells per month without anything connected? 
This makes no sence to me. Why should I store it in the BMS that is offline? 

2) The "Self-discharge" rate from the cells per month with only the BMS connected, but all input and output = 0
Makes more sense, then BMS can calulate how long the attached battery can stay in a stable state, till it gets critically damaged because of a discharge under voltage. But this value is more useful for the Appl. and not really for the BMS. 
Because the BMS see the act. cell voltage and starts to beep or blink if it see a value gets critical. (if it has this ability)

Better usage could be: In the appl. you can show a value in days (or months) if you know the act. capacity and the approx. discharge rate in % per day or per month you enter here. 

Kind regards
Ralf 

PS: looks like I not want to change the default of 0.2  atm. any longer  as long as I not know for what it is. :)