smagicld/xiaoxiangBMS #82: BMS hardware required

magnum050769, 2020-06-22:
I understand you do not sell and are in no way affiliated with the BMS hardware manufacturer.  From what I’ve understood, the iOS App only works with one manufacturer and further only two specific model numbers from that specific manufacturer; namely model numbers: JBD-SP15S001 for Li Ion and JBD-SP05S002 for LiPo.  Do I have the correct understanding or does the App work with a broader set of BMS hardware?  Also, it’s not clear where I can purchase this BMS hardware as some provided links no longer work.  Currently I am working on two projects that require 14S BMS, one rated for 60amp and the other for 100amp.  Is the BMS hardware available for purchase in the USA?   Any help, context, web links to better understand this would be greatly appreciated.

arvidtp, 2020-06-22:
I've been using this app with the BMS from [Overkill Solar](https://overkillsolar.com) for a custom LiFePO4 battery.

magnum050769, 2020-06-22:
Interesting...thanks for your reply.  I suspect the BMS provided by Overkill is actually manufactured by the same company as that for which this App was designed.  If you look at the photo of the ‘Overkill’ BMS, it’s part number is JBD-SP04S020... a very similar nomenclature as the two part numbers listed as by the developer as being compatible BMS’s.  I believe the OEM that this App was designed around is a company called LTT Power Electronic, (lithiumbatterypcb.com).  Even so, I guess we can conclude that the App at least works for other BMS part numbers from this company.

thanarious, 2020-06-23:
I am using this app with a bunch of BMSs (2x7S, 1x11S), purchased directly from lithiumbatterypcb through their aliexpress shop.

magnum050769, 2020-06-23:
Thank you - I appreciate you sharing your experience.  How do feel about the overall quality, functionality and reliability of the BMS hardware that’s compatible with this App?  I like the mobile connectivity the App brings to the hardware, but don’t want to risk having a potentially serious issue because of faulty BMS hardware.

TheDodgersOnline, 2020-07-01:
> I understand you do not sell and are in no way affiliated with the BMS hardware manufacturer. From what I’ve understood, the iOS App only works with one manufacturer and further only two specific [...]

I buy this from the retailer on AliBaba - they do give excellent service and amazing delivery times. Fromorder to my home in Australia in less than 10 days. Here is a link to the site https://3ainternational.en.alibaba.com/?spm=a2700.12243863.0.0.207a3e5fZsDvtr

Regards

closed 2021-06-13 by smagicld