smagicld/xiaoxiangBMS #83: Missing version V2.0.0

TheDodgersOnline, 2020-07-01:
I am making up a few batteries using this BMS and Bluetooth App and have been told by the seller to download the "new" app but I cannot find it. The latest version in the App Store is V1.2.16 (which I have installed) but the Release Notes mention V2.0.0 which I cannot find anywhere? Any assistance gratefully accepted

Thanks

smagicld, 2020-07-01:
The 2.0 version is not yet released.

if you want to test it, drop me your email, I’ll add you to the tester group.

TheDodgersOnline, 2020-07-01:
Hi

That would be great thanks. Do you have an estimated date yet?

Regards.

Sent from myMail for iOS

jkuo2, 2020-07-02:
> The 2.0 version is not yet released. if you want to test it, drop me your email, I’ll add you to the tester group.
> […](#)

Can you please email me the beta version? I bought the iPhone version but I no longer use iPhone. However, I have a Mac which I hope to use since this is the best app out there. 

Thank you, 

email: jaykuo10@yahoo.com

TheDodgersOnline, 2020-08-30:
Has Version 2 been released yet? If so where can I get a copy?