smagicld/xiaoxiangBMS #84: can‘t change discharge rate on IOS set to german locale

BreitiDE, 2020-07-01:
I can’t change the value of the setting, it shows 0.2 but changing just to 2 to 4,  does not take the 4. I have the language settings in IOS set to german, so trying 0,4 works to enter, but fail to save .... this might be a locale setting bug :)

smagicld, 2020-07-01:
Komisch, habe gerade versucht und es funktioniert. 

Welche Version verwendest du?

Mach doch ein screen recording damit ich es besser reproduzieren kann.

BreitiDE, 2020-07-02:
Version 1.2.17 ... hab Dir das Video geschickt. Ich könnte ja ganze Zahlen nehmen ... aber *lach*

RaWolt, 2020-11-09:
Hi,
hab das gleiche Problem. Okay das Feld ist nicht wirklich wichtig. Die Selbstendladungsrate meines LiFePo4 soll wohl nur 0.1 sein. Der Default ist 0.2 in der App. Versuche ich das umzustellen, kann ich nicht speichern. Das Problem ich habe am iPhone mit deutscher "Local" Dezimaltrenner = Komma und nicht einen Punkt.
Ich kann zwar die 2 von der 0.2 weglöschen, er lässt mich aber keine 1 tippen und stört sich an dem Punkt. 
Selbst wenn ich nicht die Nummern Tastatur nehme und aus einer anderen App "Notizen" die "0.1" einfach kopiere kommt es nicht an im Feld.   0,1 also mit Komma abspeichern geht auch nicht.  --> Werteprüfung für "Discharge Rate" Value is less or equal 0  

Meine Version ist:  IOS  V1.2.19 Build 5328

Grüße
Ralf