smagicld/xiaoxiangBMS #87: App does not connect to BMS

PhillipLabry, 2020-07-24:
I have tried this app with two different IOS devices with no luck. The BMS doesn't even show up when searching for devices. I have been able to use the android app to monitor the BMS so I know that the BMS and bluetooth work on this pack.

The app says it's version 2.0.0 but the app store says the current version is 1.2.17

Can I get an earlier version? Maybe 1.2.16 ?

smagicld, 2020-07-25:
Which BMS do you have? Send pictures / hardware details and screenshot of the android app. 

There are plenty of people trying to use the XiaoxiangBMS app with other brand BMS which naturally will not work...

PhillipLabry, 2020-07-25:
This is with the BMS sold by OverkillSolar.com

The pack is installed so I can really look at the BMS

PhillipLabry, 2020-07-25:
I mean I can not look at the BMS without taking it out of service. And I
can't take it out if service for several more weeks.

PhillipLabry, 2020-07-26:
Ok I don't know what happened but suddenly the app is working in my iPad.

If I encounter the problem again and can notice a pattern I will report it.

According to the app the BMS is manufactured by DGJBD and is a
JBD-SP04S020-L4S-129A-B-U

closed 2020-08-30 by smagicld