smagicld/xiaoxiangBMS #88: Is it possible to reset or edit the cycle count?

shinoka41, 2020-08-02:
Hi.All.
I using SP04S005.
It is 2.5v to the default of "cell minimum voltage setting" of this BMS. 
I forgot to fix it and actually dropped it to around 2.5v.
I noticed the deep discharge and charged it immediately, but the BMS's cycle count became "65513".  
Possibly due to low voltage malfunction.

I think this number of cycles is written in BMS. Is it possible to reset or edit the number of times? 
 If can't do it, I want you to do it.ASAP

smagicld, 2020-08-02:
It can not be resetted from SW. The app reads only the data.

what you could try it to disconnect the BMS, wait some time (10min?) and re-connect it. on my BMS’s it resets the counters.

TheDodgersOnline, 2020-08-15:
Hi
I was told that you can only reset the counter from a computer, not the app.
Regards

closed 2020-09-18 by smagicld