smagicld/xiaoxiangBMS #93: Word ‘Balancing’ overwrites cell voltage blue bars

aaardvaarkau, 2020-08-18:
On my iPhone 8+ in vertical format, sometimes the blue bars that show cell voltage in graphic form will extend to the right into the area where “Balancing” is written. So during balancing you don’t see useful blue bars. After balancing stops, the blue bar is not rewritten so it stays broken, see screen cap. The blue bars should stay out of the “Balancing” area (which it did when I first started the app).
[image]

CSYGD, 2020-09-21:
Hi! In other languages the word can even be longer. Maybe simply remplace the word by an arrow or other another symbol can be enough to understand the balancing process 😀

aaardvaarkau, 2020-09-22:
I need to rephrase this.  The problem seems to be that the blue balancing bars sometimes extent too far to the right.  Sometimes all of them, sometimes only one.  Above pic shows three.  It means you can no longer get good indication of relative cell voltages at a quick glance, the 'bar graph' is useless. 

It can be fixed this by disconnecting from bluetooth and reconnecting, but on my system sooner or later it will happen again.

smagicld, 2020-09-22:
Does it happen if you come back from horizontal view? (you rotate the phone 90 degrees and then back)

aaardvaarkau, 2020-09-22:
Good point. I had thought it was unpredictable, but I can reliably
reproduce the following on my iPhone 8+...

1/ Connect Bluetooth and view main screen in vertical view (portrait mode) >
everything looks good (4 “short” bars on my 4S system).
2/ turn to horizontal view > everything looks good
3/ turn back to vertical view > now have three long bars and one short bar
> stays like this until...
4/ Devices (back) and reconnect to XiaoxiangBMS BMS > starts again, all
good as in 1/.