smagicld/xiaoxiangBMS #94: Feature Request:

DaveTM75, 2020-08-18:
Dashboard view for multiple BMS’s

Hello there

Firstly - thank you for developing this. I paid for the upgrade as I think your efforts need rewarding!!

Ok, so I have 3 battery packs I manage at any one time. Being able to see SOC, ‘status’ (discharging, charging, balancing, idle, etc) and power draw data from all BMS’s on a single screen would be fabulous!

I think - on an iPhone, you could fit no more than say 5 on one page (Possibly more for larger screens, but one appreciates you have to code this) - but this would be sufficient for most people.

If there is a Bluetooth problem binding to more than one device at a time - that is fine... It would also be sufficient to cycle through each BMS to obtain the data. It would not be real-time, but it would still serve the purpose of having a relatively current dashboard view.

Please consider!

Thanks again

smagicld, 2020-08-24:
The version 2.0.0 will support it.
drop me your email, I'll add you to test users.

[image]

DaveTM75, 2020-08-24:
That’s perfect - excellent!

Do you have a publicly listed email address I can send my details to please?

Thanks

smagicld, 2020-09-18:
https://testflight.apple.com/join/HIiG2IEr - if you want to test.

closed 2020-09-18 by smagicld