smagicld/xiaoxiangBMS #98: Feature preview: Logging into CSV file (2.0.0)

smagicld, 2020-08-30:
In the 2.0 version I've added the CSV logging function.
Going to BMS details you can start recording of data into CSV file.
The file can be opened directly from mac (in finder choose your phone) or directly form the app (shared or opened in for example: apple numbers app).

Logged are only dynamic data to save the space on the device.

Post your email here if you want to participate in the testflight tests.

[image]

[image]

TheDodgersOnline, 2020-08-30:
rdodd9@icloud.com

smagicld, 2020-08-31:
ok, I've added you into the test group. as soon apple will release it - you'll get an email.

The v2 is not yet released officially. I would like to add another BMS screen (Skin). then it will be released.

dogboyaa1, 2020-09-01:
Hi I would like to be added huseinabdul1@gmail.com

Thank you for your great app!

smagicld, 2020-09-18:
https://testflight.apple.com/join/HIiG2IEr - here ist the link to 2.0.2 for all others interested with testing

CSYGD, 2020-09-21:
Hi! Thanks. Installed. I will test this function during the next days :) iOS 13.6.1 on iPhoneX. 

My dream will be to be able to log everything about the battery from the moment I welded the 18650s to the end of its life. Ideally not requiring a phone connected to BT. With an Arduino and microSD I am used to log data (temperature and humidity for my home) but I don't know how to communicate with the BMS yes. Will try to check this in the next weeks. Congrats for your work and v2.

aaardvaarkau, 2020-09-22:
Does it log only while bluetooth is connected?  If so what does it do about the gaps?

smagicld, 2020-09-22:
> Does it log only while bluetooth is connected? If so what does it do about the gaps?

Logging can be done only when connection to bms exist. No connection == no data == gaps. 

You can not fill the missing data....

jkuo2, 2020-09-22:
can you include me in the beta testing? I am using a MacBook Air right now. jaykuo10@yahoo.com. Thank you.

CSYGD, 2020-09-23:
Hi!

Two days ago I tested it during a charge. Working well on iOS13 but only when I was in the same room so the interest is limited but it is still interesting. 
I will try to check how to add an Arduino to log the all life of the bat on a SD card. If others are interest by this let me know. 
[image]

This morning I did 80km with my ebike and wanted to log the all journey but yesterday I switch to iOS14 and now it looks like it is not working anymore. Every second it tries to reconnect. Check this video screenshot:
[https://photos.app.goo.gl/EXZYrekP75BbALds7](https://photos.app.goo.gl/EXZYrekP75BbALds7)

BTW, might be gorgeous to have for the app the new iOS14 widget!

Cheers!

smagicld, 2020-09-23:
Hi

Thanks for your report. 
The data is logged only when connected to bms. The BMS itself has no buffer with past data. 

I know the big on iOS 14. Unfortunately apple has introduced lot of bugs with this release, reported during beta but not fixed yet :/. 
The one I’ve reporter is: https://developer.apple.com/forums/thread/655338

Just asking myself to wait for apple or fix it with an workaround....

aaardvaarkau, 2020-09-24:
Hi. I think it would be very useful in some circumstances to have this
logging function, but point out two reasons it might not be worth the
trouble. Not trying to be a wet blanket as I would love to see a successful
solution, but maybe this is not the right tool. . .

1) the arduino or similar device would draw current over a long period
of time that would drain the battery to greater or lesser amount, could be
much more than the BMS and Bluetooth module alone. Might be able to reduce
this by using sleep mode most of the time. This extra current draw would
not be a problem in short time recordings eg. Riding a bike for a few
hours, but would be a problem for batteries while on standby or in storage.

2) more difficult I believe is that the BMS is not able to detect currents
less than 200 or 300mA. This means some loads and some low current chargers
would be missed entirely and your logging could be very inaccurate.

I think you’d need a different BMS to make this work.  (Actually what I
just said applies to my 120A BMS, but lower current models might work ok. I
haven’t seen any input about low current BMSs from LLT. )

JR

TheDodgersOnline, 2020-10-04:
Hi - when I try this on my iPad the data is collected but when I try to open the file the app crashes. I have reported this via the app but do not know if this is being addressed.
Regards

ulihuber, 2021-06-06:
uli@uhu-servo.de

Thanks :-)

smagicld, 2026-10-19:
hi @markgol 
can you provide steps and screenshots to reproduce?
just tested on latest ios 27. i can save and open config files. it works without any issue. please note that before you open something - you have to have saved something otherwise the list is empty.

markgol, 2026-10-19:
Thank you for the very prompt response.

I am specifically referring to the recording .csv file. I looks like it successfully is recording data.  I then stop recording and click Open or click Open while recording  the app appears to go into the background and then nothing happens.  I can not find that the .csv file is accessible in 'Files' either on iCloud or local.  I do not see any options to save or copy the file to a folder.
'Delete' also appears to function incorrectly.  It shows the .csv as empty.
I have attached 3 screenshots.
Maybe I am not doing something correctly or missed something in the settings.

[image]
[image]
[image]

smagicld, 2026-10-19:
Hi @markgol thanks for your Input. I was thinking about xioaxiang App and opening / saving of configs. But your issue is related to the BMSTool app. I’ll check it during the day.

smagicld, 2026-10-20:
Hi @markgol I was able to reproduce your issue on ipad and I think i could fix it. i will submit new release now to apple. if you want to test it before apple release - drop me your email and I'll add you to testflight.

markgol, 2026-10-22:
Thank you for the pompt response.  Policies on my Apple only let me 
update or install through Apple app store.
sincerely,
Mark