smagicld/xiaoxiangBMS #99: Pro Version is not working "BMS data download" with LIONTRON Battery from 29 June 2020

ulixWOMO, 2020-09-04:
I use the PRO Version (German language), With my allready in use LIONTRON Battery manufctured in January 2020 i have no problem to downlad BMS data. But with my new battery from June 2020 i get allways an Error : Fehler keine bestätigungvom BMS Versuche noch einmal

smagicld, 2020-09-04:
Hi

Sorry, this error message does not exist in the App.
Pleas provide App version / Build number, phone type / iOS version and some screenshots.

ulixWOMO, 2020-09-04:
Hi,

I’m using iPhone 11 Pro Max , IOS 13.7 and Xiaoxiang BMS V. 1.2.18

Atached you find a screenshot

Kind regards

Ulrich Sigmann

smagicld, 2020-09-04:
there is no screenshot

ulixWOMO, 2020-09-04:
For better explanation a screen recording

smagicld, 2020-09-04:
sorry, nothing images / videos. please upload them directly on github.

ulixWOMO, 2020-09-04:
[vlc-record-2020-09-04-10h34m19s-out_2020-09-04-10-01-36.MP4-.zip](https://github.com/smagicld/xiaoxiangBMS/files/5173629/vlc-record-2020-09-04-10h34m19s-out_2020-09-04-10-01-36.MP4-.zip)

smagicld, 2020-09-04:
thanks. this helps a lot. and the BMS Screen - do you have there some data? please post the screenshot.

ulixWOMO, 2020-09-04:
[image]
Which aditional data do you need?

smagicld, 2020-09-04:
it would be good to habe the video (the zip has an empty mpg file). please send it to my mail: lukas at deszynski dot com.

smagicld, 2020-09-18:
as discussed, without having the HW can not support it.

closed 2020-09-18 by smagicld