smagicld/xiaoxiangBMS #100: I want to know a lot about smart BMS commands.

nekosoft2000, 2020-09-13:
I came to want to study smart BMS.
Currently, I am making a BMS for LEV40 with reference to Outlander PHEV BMS.
I want to use this app around the user interface, so I'd be happy if I could exchange information.

The Ver2.0 app is great, so I'm looking forward to an early release.
If possible, I would like to participate in the beta test.

http://nekosoftpc.blogspot.com/2020/09/bmslcd.html

smagicld, 2020-09-18:
Hi. https://www.lithiumbatterypcb.com/smart-bms-software-download/ there you'll find all infos you need.

closed 2020-09-18 by smagicld

nekosoft2000, 2020-09-29:
Thank you for your valuable information. I was able to make BMS safely.

[https://youtu.be/102elER1Yao](https://youtu.be/102elER1Yao)