smagicld/xiaoxiangBMS #101: In-app purchases get disabled after app reboot

MentalPower, 2020-09-13:
Hi,

I've noticed that the app constantly needs to restore purchases whenever it's restarted. I'm going to use my BMS where there's not a stable connection, which makes this behavior undesirable. Also, other apps don't reset their IAPs on every reboot but rather authenticate the AppStore receipt once and store that confirmation locally such that it survives app restarts. I can look into fixing that if you'll accept a pull request. Otherwise, consider this a bug report.

Thanks!

smagicld, 2020-09-14:
HI
Issue can not be reproduced.
Which version of app, iOS and device are you using?
describe necessary steps / video to reproduce.

/lukas

MentalPower, 2020-09-14:
Repro is super easy.

1. Force quit the app
2. Re-open app
3. Restore purchases via the info menu on the bottom right.
4. Force quit app again.
5. iAP gets reset

I’m on iOS 14.0 (18A5373a).

[image]

smagicld, 2020-09-14:
You have this problem because you're using iOS14 beta. It has still lot of bugs.
IOS <= 13.7 works.

closed 2020-09-14 by smagicld

MentalPower, 2020-09-14:
iOS 14 will very likely go GA on Tuesday. I’d double check your code :).

MentalPower, 2020-09-14:
iOS14 introduced some changes to how iAPs work. These two sessions can prove instructive:

1. https://developer.apple.com/videos/play/wwdc2020/10661
2. https://developer.apple.com/videos/play/wwdc2020/10659

smagicld, 2020-09-15:
thanks for your goodwill, but the problem is not in iAP. It's related with some encoding functions on the beta. until GM I usually do not fix anything because the betas are very unstable.

MentalPower, 2020-09-16:
GM is out today and this is still an issue.