smagicld/xiaoxiangBMS #102: Xiaoxiang App

WS8, 2020-09-16:
I have the Xiaoxiang BMS App for iPhone (pro version).
Is there way to send the the readout to a Mac/PC?
Or is there any version for Mac/PC available?

smagicld, 2020-09-16:
Yes, in v2. Drop me your email - I will add you to the test users.

WS8, 2020-09-16:
Hi,

Thanks for adding me to the test users for V2.
What do I need to do to get that vetsion?

Thanks!
Wim

smagicld, 2020-09-18:
https://testflight.apple.com/join/HIiG2IEr here the link for all others to test this build.

closed 2020-09-18 by smagicld

yourob02, 2022-06-20:
> Yes, in v2. Drop me your email - I will add you to the test users. 
> 
> Von meinem iPhone gesendet
> 
> > 
> > 
> > I have the Xiaoxiang BMS App for iPhone (pro version).
> > Is there way to send the the readout to a Mac/PC?
> > Or is there any version for Mac/PC available?

Robyoung90@icloud.con