smagicld/xiaoxiangBMS #105: Balance while Charging and not charging

mmaattppllaayy, 2020-10-04:
Please add a setting that allows the bms to balance when charging and when not charging. I want it to balance all the time.

TheDodgersOnline, 2020-10-04:
Hi - this already exists on the IOS version.

TheDodgersOnline, 2020-10-04:
![Uploading 95028738...]

TheDodgersOnline, 2020-10-04:
[image]

mmaattppllaayy, 2020-10-04:
The balancer enabled only lets the balancer come on when the other setting determines. It doesn’t enable balancing all the time.

mmaattppllaayy, 2020-10-06:
Any update on adding that setting?

smagicld, 2020-10-06:
As other already wrote, this is already implemented.

closed 2020-10-06 by smagicld

mmaattppllaayy, 2020-10-06:
This is not already available!!! The balancer enabled function only allows the balancer to be on when the second setting of balance only while charging is enabled. 
When I turn balance while charging off this allows it to balance but then when charging it doesn’t balance. 
Then when I turn balance while charging on this makes it balance when charging but not when not charging. 
It is one or the other not both. I need both.